hatem-darweesh/op_bridge #4: Question about run run_exploration_mode_ros2.sh

gaokao123, 2022-09-22:
I follow the CARLA Simulator + Autoware Universe Tutorials in URL of https://github.com/autowarefoundation/autoware/discussions/2828 that you released, but there are some errors,do you have any idea? thank you！
[image]

Another question,  can the environment variables replace  "export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py2.7-linux-x86_64.egghuanj" with "export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py3.7-linux-x86_64.egghuanj"

hatem-darweesh, 2022-09-25:
I think you have solved this issue already.

closed 2022-09-25 by hatem-darweesh

qiunai, 2023-10-18:
> yes, I have solved this issue already, thank you!

How did you solve this problem? I'm currently facing the same issue
I'm glad to hear from you, thank you！

Traceback (most recent call last):
  File "/home/auto/workspaces/autowarefoundation/op_bridge/op_bridge/op_bridge_ros2.py", line 61, in _tick_agent
    ego_action = self.agent()
  File "/home/auto/workspaces/autowarefoundation/op_bridge/leaderboard/autoagents/agent_wrapper.py", line 75, in __call__
    return self._agent()
  File "/home/auto/workspaces/autowarefoundation/op_bridge/leaderboard/autoagents/autonomous_agent.py", line 115, in __call__
    control = self.run_step(input_data, timestamp)
  File "/home/auto/workspaces/autowarefoundation/op_bridge/op_bridge/op_ros2_agent.py", line 531, in run_step
    raise RuntimeError("Stack exited with: {} {}".format(
RuntimeError: Stack exited with: 1 None

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/auto/workspaces/autowarefoundation/op_bridge/op_bridge/op_bridge_ros2.py", line 171, in run_agent
    self.agent_loop._tick_agent(timestamp)
  File "/home/auto/workspaces/autowarefoundation/op_bridge/op_bridge/op_bridge_ros2.py", line 67, in _tick_agent
    raise AgentError(e)
leaderboard.autoagents.agent_wrapper.AgentError: Stack exited with: 1 None
Scenario Ended , Hero has fallen, Sayonara
ERROR: failed to destroy actor 676 : unable to destroy actor: not found

Desperadozcx, 2024-09-20:
> How did you solve this problem? I'm currently facing the same issue I'm glad to hear from you, thank you！
>
> Traceback (most recent call last): File [...]

I encountered the same problem as you. Have you resolved this issue now?