hatem-darweesh/op_bridge #6: ERROR when run_exploration_mode_ros2.sh

CedarTH, 2022-10-08:
I want to use ros2 so I run run_exploration_mode_ros2.sh.However I meet following error.I have tried changing default python version to python2.7 but it's still not working.

Traceback (most recent call last):
  File "/home/tasl/carla-autoware-universe/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 12, in <module>
    import carla
  File "<frozen zipimport>", line 259, in load_module
  File "/home/tasl/carla_0.9.13/PythonAPI/carla/dist/carla-0.9.13-py2.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
    raise ImportError('This package shoud not be accessible on Python 3.')
  File "<frozen zipimport>", line 259, in load_module
  File "/home/tasl/carla_0.9.13/PythonAPI/carla/dist/carla-0.9.13-py2.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "/home/tasl/carla_0.9.13/PythonAPI/carla/dist/carla-0.9.13-py2.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
  File "/usr/lib/python3.8/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: /home/tasl/.cache/Python-Eggs/carla-0.9.13-py2.7-linux-x86_64.egg-tmp/carla/[libcarla.so](http://libcarla.so/): undefined symbol: _PyThreadState_Current

hatem-darweesh, 2022-10-08:
Please review the installation instruction and tutorials. 
Python version is an annoying problem. You can find better guidance in CARLA discord.

CedarTH, 2022-10-09:
I am sorry that I'm still confused.

I have watched your video "CARLA-Autoware universe Tutorial" in youtube and successfully install autoware.universe,and  successfully run autoware.universe in Planning simulation.
However,I want to use CARLA as my simulator.However,I'm not sure which version of python should be used so I meet above problem.I have tried following ways to run run_exploation_mode_ros2.sh :

1.(default)use Python3.7:meet error "no module named 'autoware_auto_vehicle_msgs'
2.(change python version)use Python2.7:meet errer caused by the conflct betweent python2 and python3,such as "from queue import Queue" "no module named queue"

Is there a tutorial video which help me to successfully "OP_bridge"?I have checked all your videos in youtube but not found. It seemd that there is only "op bridge ros2 test run".
Thank you very much!

msanchezvicom, 2022-10-11:
Hi @hatem-darweesh  , 
your work is amazing, I am very impressed.
Unfortunately I can't test the integration between Autoware universe and CARLA since I am facing the same problem as @CedarTH  . 
@CedarTH have you finally solved it, do you have any other idea that can lead us to the solution?
Thanks

yuting-fu, 2022-10-12:
> 
> 1.(default)use Python3.7:meet error "no module named 'autoware_auto_vehicle_msgs' 

A quick solution could be installing the `autoware_auto_vehicle_msgs` in your ros2 environment.

AviBen96, 2023-02-15:
I am actually getting to the same point Where autoware and Rviz open but the Initializing state never ends . 

[scenario_selector-59] [INFO] [1676466881.162732721] [planning.scenario_planning.scenario_selector]: Waiting for route.
[component_container_mt-74] [WARN] [1676466881.163323666] [autoware_api.external.vehicle_status]: The turn_indicators topic is not subscribed
[component_container-67] [INFO] [1676466884.906256824] [control.trajectory_follower.lane_departure_checker_node]: waiting for route msg...
[component_container-67] [WARN] [1676466885.029031545] [control.operation_mode_transition_manager]: Engage unavailable: trajectory size must be > 2
[system_error_monitor-21] [ERROR] [1676466885.029823853] [system_error_monitor /autoware/control/autonomous_driving/node_alive_monitoring]: [Single Point Fault]: Error
[system_error_monitor-21] [ERROR] [1676466885.029851026] [system_error_monitor /autoware/control/autonomous_driving/node_alive_monitoring/topic_status/topic_state_monitor_control_command_control_cmd: control_topic_status]: [Single Point Fault]: Error
[system_error_monitor-21] [ERROR] [1676466885.029857230] [system_error_monitor /autoware/control/autonomous_driving/node_alive_monitoring/topic_status/topic_state_monitor_trajectory_follower_control_cmd: control_topic_status]: [Single Point Fault]: Error
[system_error_monitor-21] [ERROR] [1676466885.029861283] [system_error_monitor /autoware/localization/performance_monitoring/localization_accuracy]: [Single Point Fault]: Error
[system_error_monitor-21] [ERROR] [1676466885.029864862] [system_error_monitor /autoware/localization/performance_monitoring/localization_accuracy/localization_error_monitor: localization_accuracy]: [Single Point Fault]: ellipse size is over the expected range
[system_error_monitor-21] [ERROR] [1676466885.029869362] [system_error_monitor /autoware/localization/performance_monitoring/localization_accuracy/localization_error_monitor: localization_accuracy_lateral_direction]: [Single Point Fault]: ellipse size along lateral direction is over the expected range
[system_error_monitor-21] [ERROR] [1676466885.029873821] [system_error_monitor /autoware/perception/node_alive_monitoring]: [Single Point Fault]: Error
[system_error_monitor-21] [ERROR] [1676466885.029877067] [system_error_monitor /autoware/perception/node_alive_monitoring/topic_status/topic_state_monitor_object_recognition_objects: perception_topic_status]: [Single Point Fault]: OK
[system_error_monitor-21] [ERROR] [1676466885.029880762] [system_error_monitor /autoware/perception/node_alive_monitoring/topic_status/topic_state_monitor_obstacle_segmentation_pointcloud: perception_topic_status]: [Single Point Fault]: Error
[system_error_monitor-21] [ERROR] [1676466885.029885135] [system_error_monitor /autoware/planning/node_alive_monitoring]: [Single Point Fault]: Error
[system_error_monitor-21] [ERROR] [1676466885.029888190] [system_error_monitor /autoware/planning/node_alive_monitoring/topic_status/topic_state_monitor_mission_planning_route: planning_topic_status]: [Single Point Fault]: Error
[system_error_monitor-21] [ERROR] [1676466885.029891424] [system_error_monitor /autoware/planning/node_alive_monitoring/topic_status/topic_state_monitor_scenario_planning_trajectory: planning_topic_status]: [Single Point Fault]: Error
[component_container_mt-64] [WARN] [1676466885.890581577] [planning.scenario_planning.lane_driving.motion_planning.surround_obstacle_checker]: waiting for pointcloud info...
[component_container_mt-62] [INFO] [1676466886.443339284] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner]: waiting for scenario_topic

biubiubiu919, 2023-08-03:
> > Can you tell me what is your target ? CARLA + autoware.universe ? CARLA + autoware.ai ? there are different branches for each case and different tutorial for each case and different os, ros, python requirements
> 
> Hello @CedarTH, @hatem-darweesh :
> 
> My target is CARLA 9.0.13 + Autoware.Universe. I could finally solve the issue by modifying the CARLA egg file export from _export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py2.7-linux-x86_64.egg_ to **export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py3.7-linux-x86_64.egg**
> 
> It is important to note that the PYTHONPATH variable should be previously cleaned up: **export PYTHONPATH=""** export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/util export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/agents **export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py3.7-linux-x86_64.egg**
> 
> However, the AutowareStatePanel state stays forever in _Initializing_ mode. I have tried to relaunch Autoware several times but unfortunateluy I cannot make it work.
> 
> ![image](https://user-images.githubusercontent.com/112627154/195535012-c688e42e-db22-4a99-80cb-4eb4561f1a

hello @ msanchezvicom 
Did you solve this problem, i seem to encouter the same problem. the state stuck in Initializing mode

msanchezvicom, 2023-08-03:
Hi @biubiubiu919 ,
After realizing what I stated, I was able to solve the problem by reinstalling/building everything, following the steps diligently.
BR