hatem-darweesh/op_bridge #7: An error with launching "run_exploration_mode_ros2.sh"

naokiadachi0219, 2022-10-11:
Hi, I am in trouble with an error with launching "run_exploration_mode_ros2.sh" .

The operation environment is as follow:
1.Autoware PC
  OS : Ubuntu 20.04
  CPU : Intel Core i9
  RMA : 32Gbytes
  GPU : NVIDIA GeForce RTX2060 SUPER
2.CARLA PC
  OS : Ubuntu 18.04

[image]
[image]

I would appreciate your cooperation.
Thanks in advance.

yuting-fu, 2022-10-12:
I ran into the same error. My workaround was to comment out line 530-532 in the script `op_ros2_agent.py`:
```
# check if stack is still running
# if self.stack_process and self.stack_process.poll() is not None:
#        raise RuntimeError("Stack exited with: {} {}".format(
#        self.stack_process.returncode, self.stack_process.communicate()[0]))
```

naokiadachi0219, 2022-10-18:
@yuting-fu 
Thanks for your advice.
I tried. I didn't get any errors, but the process never ended and nothing happened.

yuting-fu, 2022-10-19:
> @yuting-fu Thanks for your advice. I tried. I didn't get any errors, but the process never ended and nothing happened.

Did you also launch autoware using the launch file or script stated in Hatem's tutorial? The `op_ros2_agent.py` is only the agent between CARLA simulator and Autoware. The autoware modules (perception, planning, control, etc.) have to be launched in ROS2.

If everything is running properly, you should be able to manually send a routing request in RVIZ and see the car moving automatically along the planned trajectory, just like what Hatem showed in the tutorial video. The autoware might get stuck in initializing state in RVIZ. Then just relaunch everything a couple of times and it will work.

Good Luck.

naokiadachi0219, 2022-11-10:
For everyone who gave me advice

I was able to solve the problem and run.
Thank you.

I changed python version from 3.8 to 3.7.

If I get stuck again, please help me.
Thank you.

closed 2022-11-10 by naokiadachi0219

marcusvinicius178, 2022-11-18:
Hi @naokiadachi0219 I am facing the same issue could you please take a quick look at my issue: https://github.com/hatem-darweesh/op_bridge/issues/8#issuecomment-1319258106

I had the same issues as you with Python 3.8. However I have accidentally removed it from my system and broke Ubuntu! 

I would like to know how did you use Python3.7? In a virtual environment? Could you pass the tutorial you have installed it? And the steps which you took to use successfully to run the script?

Thanks in advance

naokiadachi0219, 2022-11-21:
Hi @marcusvinicius178 

Sorry.
I may not be able to meet your expectations.

First , I deployed without using a virtual machine.

I used this tutorial.
https://www.youtube.com/watch?v=EFH-vVxn180
https://www.youtube.com/watch?v=dxwwNacez7o
However, the change in the version is not mentioned in the video.
I changed the version on my own.
Installed in Python3.7,The script is executed in Python3.8.

marcusvinicius178, 2022-11-22:
@naokiadachi0219 no worries I was able to finally do the integration. I have just one question.

Which Nvidia-driver and Cuda version did you use?

Because I am having some issues with Rviz to visualize the camera data, and also some warning regarding my system compatibility  (driver,etc)

naokiadachi0219, 2022-11-28:
@marcusvinicius178

Thank you for waiting.
Here are the versions:

+-----------------------------------------------------------------------------+
| NVIDIA-SMI 520.61.05    Driver Version: 520.61.05    CUDA Version: 11.8     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:02:00.0  On |                  N/A |
| 37%   37C    P8     7W / 175W |    308MiB /  8192MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
I hope this helps you.