hatem-darweesh/op_bridge #9: how to fix the direction

yironglin, 2022-12-26:
Hello, I followed the instructions and run run_exploration_mode_ros2.sh. I managed to start the autoware, however it seemed like the directions of the vehicle are different in carla and rviz, like this:
[image]
Any hint to fix the problem? I have already modified the ~/src/launcher/autoware_launch/autoware_launch/launch/autoware.launch.xml and ~/src/sensor_kit/sample_sensor_kit_launch/sample_sensor_kit_launch/launch/gnss.launch.xml, did I miss other files?

hatem-darweesh, 2022-12-26:
check this
https://github.com/ZATiTech/open_planner/issues/2#issuecomment-1256142611

yironglin, 2022-12-26:
@hatem-darweesh Thank you for your help! I modified these 2 files then all things work properly!