hatem-darweesh/op_bridge #10: How to fix this error

HyunkiKim0407, 2023-01-04:
### Hi, when i ran exploration_made_ros2.sh gave this error i don't know how fix it ###

a@a:~/op_carla/op_bridge/op_scripts(ros2)$ ./run_exploration_mode_ros2.sh
Ego Vehicle:  hero
Executing stack... hero Town01
-------------------------------
Starting OpenPlanner ..
hero
Town01
true

-------------------------------
[INFO] [launch]: All log files can be found below /home/a/.ros/log/2023-01-04-15-50-48-939164-a-227460
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure("executed command failed. Command: xacro /home/a/carla-autoware-universe/autoware.universe.openplanner/install/tier4_vehicle_launch/share/tier4_vehicle_launch/urdf/vehicle.xacro vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit config_dir:=/home/a/carla-autoware-universe/autoware.universe.openplanner/install/sample_sensor_kit_description/share/sample_sensor_kit_description/config\nCaptured stderr output: error: name 'xacro' is not defined \nwhen evaluating expression 'xacro.load_yaml('$(find sample_vehicle_description)/config/vehicle_info.param.yaml')' \nwhen evaluating expression 'vehicle_info['/**']['ros__parameters']['wheel_base']/2.0'\nwhen processing file: /home/a/carla-autoware-universe/autoware.universe.openplanner/install/sample_vehicle_description/share/sample_vehicle_description/urdf/vehicle.xacro\nincluded from: /home/a/carla-autoware-universe/autoware.universe.openplanner/install/tier4_vehicle_launch/share/tier4_vehicle_launch/urdf/vehicle.xacro\n")>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 5 more times]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/actions/node.py", line 465, in execute
    self._perform_substitutions(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/actions/node.py", line 420, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 164, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 133, in evaluate_parameter_dict
    evaluated_value = value.evaluate(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/parameter_descriptions.py", line 91, in evaluate
    self.__evaluated_parameter_value = perform_typed_substitution(
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/type_utils.py", line 546, in perform_typed_substitution
    perform_substitutions(context, cast(List[Substitution], value)),
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/substitutions/command.py", line 119, in perform
    raise SubstitutionFailure(on_error_message)
launch.substitutions.substitution_failure.SubstitutionFailure: executed command failed. Command: xacro /home/a/carla-autoware-universe/autoware.universe.openplanner/install/tier4_vehicle_launch/share/tier4_vehicle_launch/urdf/vehicle.xacro vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit config_dir:=/home/a/carla-autoware-universe/autoware.universe.openplanner/install/sample_sensor_kit_description/share/sample_sensor_kit_description/config
Captured stderr output: error: name 'xacro' is not defined
when evaluating expression 'xacro.load_yaml('$(find sample_vehicle_description)/config/vehicle_info.param.yaml')'
when evaluating expression 'vehicle_info['/**']['ros__parameters']['wheel_base']/2.0'
when processing file: /home/a/carla-autoware-universe/autoware.universe.openplanner/install/sample_vehicle_description/share/sample_vehicle_description/urdf/vehicle.xacro
included from: /home/a/carla-autoware-universe/autoware.universe.openplanner/install/tier4_vehicle_launch/share/tier4_vehicle_launch/urdf/vehicle.xacro

Traceback (most recent call last):
  File "/home/a/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 61, in _tick_agent
    ego_action = self.agent()
  File "/home/a/op_carla/op_bridge/leaderboard/autoagents/agent_wrapper.py", line 75, in __call__
    return self._agent()
  File "/home/a/op_carla/op_bridge/leaderboard/autoagents/autonomous_agent.py", line 115, in __call__
    control = self.run_step(input_data, timestamp)
  File "/home/a/op_carla/op_bridge/op_bridge/op_ros2_agent.py", line 531, in run_step
    raise RuntimeError("Stack exited with: {} {}".format(
RuntimeError: Stack exited with: 1 None

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/a/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 171, in run_agent
    self.agent_loop._tick_agent(timestamp)
  File "/home/a/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 67, in _tick_agent
    raise AgentError(e)
leaderboard.autoagents.agent_wrapper.AgentError: Stack exited with: 1 None
Scenario Ended , Hero has fallen, Sayonara
ERROR: failed to destroy actor 197 : unable to destroy actor: not found

hatem-darweesh, 2023-01-10:
The error source could be anything. 
Could you narrow it down?
Have you tried autoware.universe planning tutorial first. ? 
Make sure that CARLA simulator is working and try to call any command line python such as changing the environment or loading new actors. 
Make sure the paths to maps and other resources are set correctly. 
Check previous closed issues related to the CARLA autoware bridge.

HyunkiKim0407, 2023-01-11:
I checked the tutorial and followed the process.
However, in the process of modifying autoware.launch, I did not write this
code, and the autoware was executed. But it didn't work with carla and
nothing worked. When I inserted the <--carla --> code again, I got the
error "failed to destroy acotor" and I thought it was wrong somewhere.
Thank you.

AditiShelke, 2023-01-17:
It's hard to give a definitive answer without knowing more about the context and the codebase, but it seems that there is an issue in the self.agent() function that is causing it to exit with an error code of 1 and None, which is then causing the code to raise a runtime error.

HyunkiKim0407, 2023-01-26:
The error is continuing now. Looking at the tick_agent function, it seems to be related to timestamp, so I want to modify that part. If carla and autoware modify sim_time modified in autoware.launch elsewhere, will the error be fixed? If that's right, which part should I modify?

yaoweixiao-ux, 2023-03-17:
hi I meet this problem too, we using offical module, but when we change xml add Carla, err shows no xxx agent;
I solve this problem by at autoware/src/universe/external git clone hatem-darweesh open_planning module and colcon build again , launch again right

HyunkiKim0407, 2023-03-17:
Thanks comment, I'm sorry i can't found open_planning hatem-darweesh github could you give me a link please?

HyunkiKim0407, 2023-03-17:
> hi I meet this problem too, we using offical module, but when we change xml add Carla, err shows no xxx agent; I solve this problem by at autoware/src/universe/external git clone hatem-darweesh open_planning module and colcon build again , launch again right

Thanks comment, I'm sorry i can't found open_planning hatem-darweesh github could you give me a link please?

yaoweixiao-ux, 2023-03-21:
https://github.com/ZATiTech/open_planner

YimingChen-Real, 2023-03-28:
make sure in file xxx/op_agent/start_ros2.sh that map_path:=$HOME/xxx
refer to: [https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/#how-to-run-a-planning-simulation](https://github.com/hatem-darweesh/op_bridge/issues/url)