hatem-darweesh/op_bridge #12: Stuck on planning

AviBen96, 2023-02-15:
Good evening,

Fisrt of all thank you so much for your work. The youtube videaos, comments and bridge are extremely usefull ! 

I managed to make the bridge launch but the run_exploration gets stucked on planning after sending a goal. 

I get this error in loop 

[system_error_monitor-21] [ERROR] [1676488552.295627924] [system_error_monitor /autoware/perception/node_alive_monitoring/topic_status/topic_state_monitor_obstacle_segmentation_pointcloud: perception_topic_status]: [Single Point Fault]: Error
[component_container_mt-74] [WARN] [1676488552.529779971] [autoware_api.external.vehicle_status]: The turn_indicators topic is not subscribed
[component_container_mt-64] [ERROR] [1676488552.772213389] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: Failed to look up transform from map to base_link
[component_container_mt-64] [ERROR] [1676488553.360538686] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: Failed to look up transform from map to base_link
[component_container_mt-64] [ERROR] [1676488553.938515903] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: Failed to look up transform from map to base_link
[component_container_mt-64] [ERROR] [1676488554.510493746] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: Failed to look up transform from map to base_link
[component_container_mt-64] [ERROR] [1676488555.074671410] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: Failed to look up transform from map to base_link
[component_container_mt-64] [ERROR] [1676488555.653173686] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: Failed to look up transform from map to base_link
[system_error_monitor-21] [ERROR] [1676488556.124665306] [system_error_monitor /autoware/localization/performance_monitoring/localization_accuracy]: [Single Point Fault]: Error

Anyone knows what is happening and how to fix it ?

hatem-darweesh, 2023-02-18:
I had this issue to, with me it sometimes work, other time not. 
there are some suggestions at openplanner-wg channel at discord. please check them.

AviBen96, 2023-02-19:
Thank you for your answer ! I will take a look. 

Everything seems to work seperatly, but very often one thing seems to break. For example the camera often works when it gets stuck on planning / initializing, and most of the time when the pla,nning work and the car it driven the camera is broken and I get NO Image. 

I am trying to automatize some tests using the bridge, but this makes it nearly impossible since I can not forseen those bug ...

AviBen96, 2023-02-20:
I have been through the open planner channel of Autoware discord (are you referring to this, or to a specific openplanner discord community ? if so, can you provide a link, I can not find it). The only things I found there were report of errors, but no suggestions of solutions... 

Could you send a link to the suggestions you are referring to ? 

Thank you very much !

FlowEternal, 2023-03-10:
> I have been through the open planner channel of Autoware discord (are you referring to this, or to a specific openplanner discord community ? if so, can you provide a link, I can not find it). The [...]

I also get stuck in STATE PLANNING and can not change state to “wait engage”. have you already solved it? if so, how do you solve it

FlowEternal, 2023-03-11:
> I had this issue to, with me it sometimes work, other time not. 
> there are some suggestions at openplanner-wg channel at discord. please check them. 
I am also stuck in this problem. could you attach the related link here.thanks

AviBen96, 2023-03-11:
No I did not solve it. Somethings that helped for it to happen less often : kill all ros2 process before launching, kill all python process. and make sure the cpu / gpu are not in too much of a rush

FlowEternal, 2023-03-12:
> No I did not solve it. Somethings that helped for it to happen less often : kill all ros2 process before launching, kill all python process. and make sure the cpu / gpu are not in too much of a rush

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
[image]

[image]

I tried but it seems not work. Still stuck on Planning and the error Single Point Fault keep popping out
any suggestion?  may be there are some bug in the single point tracking part of control module and the config file or source code need to be adjusted

AviBen96, 2023-03-12:
Yes I often have the same problem

Eisoc, 2023-04-04:
> I had this issue to, with me it sometimes work, other time not. there are some suggestions at openplanner-wg channel at discord. please check them.

Hi, what is the name of the channel? openplanner? i can not find it..
maybe it is not public, so can you provide an invitation link?
Thank you very much