hatem-darweesh/op_bridge #13: Leaderboard training scenario, list out of range

AviBen96, 2023-02-17:
Good evening, 

I am trying to link autoware to the leaderboard training scenario. 

I get this error 

The scenario could not be loaded:
> list index out of range

Traceback (most recent call last):
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/leaderboard_evaluator.py", line 332, in _load_and_run_scenario
    scenario = RouteScenario(world=self.world, config=config, debug_mode=args.debug)
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/scenarios/route_scenario.py", line 189, in __init__
    self._update_route(world, config, debug_mode>0)
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/scenarios/route_scenario.py", line 235, in _update_route
    self.timeout = self._estimate_route_timeout()
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/scenarios/route_scenario.py", line 266, in _estimate_route_timeout
    prev_point = self.route[0][0]
IndexError: list index out of range
> Registering the route statistics
Exception ignored in: <function RouteScenario.__del__ at 0x7f094178e280>
Traceback (most recent call last):
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/scenarios/route_scenario.py", line 567, in __del__
    self.remove_all_actors()
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/scenario_runner/srunner/scenarios/basic_scenario.py", line 202, in remove_all_actors
    for i, _ in enumerate(self.other_actors):
AttributeError: 'RouteScenario' object has no attribute 'other_actors'

It works properly with the provided scripts. Am I missing something ? 

Thank you !

hatem-darweesh, 2023-02-18:
are you trying the ROS2 version with autoware.universe ?

AviBen96, 2023-02-19:
Yes indeed. I using Ubuntu 20 with ros galactic and I have everything set as explained in your videos and repos. 

The provided route seems to work, but if I try to link the bridge to the latest leaderboard bridge, it gives me this error. 

Also : Is there a way to automatically let autoware run the whole route ? Or do I have to send goals manually ?

Also, with the provided route, as soon as one route ends I get this error and it fails to load the next one. 

========= Preparing RouteScenario_2 (repetition 0) =========
> Setting up the agent

Could not set up the required agent:
> 

Traceback (most recent call last):
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/leaderboard_evaluator.py", line 285, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/autoagents/autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/op_bridge/op_ros2_agent.py", line 101, in setup
    rclpy.init(args=None)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/__init__.py", line 76, in init
    return context.init(args, domain_id=domain_id)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/context.py", line 70, in init
    raise RuntimeError
RuntimeError
> Registering the route statistics

========= Preparing RouteScenario_3 (repetition 0) =========
> Setting up the agent

Could not set up the required agent:
> 

Traceback (most recent call last):
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/leaderboard_evaluator.py", line 285, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/leaderboard/autoagents/autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "/home/avner/CARLA_AUTOWARE_project/op_carla/op_bridge/op_bridge/op_ros2_agent.py", line 101, in setup
    rclpy.init(args=None)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/__init__.py", line 76, in init
    return context.init(args, domain_id=domain_id)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/context.py", line 70, in init
    raise RuntimeError
RuntimeError

ry4nzhu, 2024-04-19:
How to get ROS2 version with autoware.universe run the leaderboard tests? It seems to me that the $route_topic is never used in the 'start_ros2.sh' from the op_agent. The ego vehicle does not move after running `run_route_scenarios_ros2.sh`. Can you guide me on how to make the leaderboard test run correctly (I'm using ROS galactic and autoware.universe)?