hatem-darweesh/op_bridge #14: issue when running ./run_exploration_mode_ros2.sh

hengcong, 2023-03-05:
hi, I get the error massage like this. I cannot get the autoware started but the carla display did change to town1
```
[INFO] [launch]: All log files can be found below /home/hguo/.ros/log/2023-03-04-22-46-05-299220-hguo-Legion-5-Pro-22552
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure("launch configuration 'launch_carla_interface' does not exist")>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 106, in visit
    if self.__condition is None or self.__condition.evaluate(context):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/condition.py", line 44, in evaluate
    return self._predicate(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/conditions/if_condition.py", line 43, in _predicate_func
    return evaluate_condition_expression(context, self.__predicate_expression)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/conditions/evaluate_condition_expression_impl.py", line 41, in evaluate_condition_expression
    expanded_expression = perform_substitutions(context, expression)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/substitutions/launch_configuration.py", line 98, in perform
    raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: launch configuration 'launch_carla_interface' does not exist
Traceback (most recent call last):
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 61, in _tick_agent
    ego_action = self.agent()
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/leaderboard/autoagents/agent_wrapper.py", line 75, in __call__
    return self._agent()
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/leaderboard/autoagents/autonomous_agent.py", line 115, in __call__
    control = self.run_step(input_data, timestamp)
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/op_bridge/op_ros2_agent.py", line 531, in run_step
    raise RuntimeError("Stack exited with: {} {}".format(
RuntimeError: Stack exited with: 1 None

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 171, in run_agent
    self.agent_loop._tick_agent(timestamp)
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 67, in _tick_agent
    raise AgentError(e)
leaderboard.autoagents.agent_wrapper.AgentError: Stack exited with: 1 None
Scenario Ended , Hero has fallen, Sayonara
ERROR: failed to destroy actor 375 : unable to destroy actor: not found 
^CException ignored in: <module 'threading' from '/usr/lib/python3.8/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 1388, in _shutdown
    lock.acquire()
TypeError: _stop_loop() takes 1 positional argument but 2 were given

```
I would appreciate if you can help me.

hengcong, 2023-03-07:
I revise

hengcong, 2023-03-07:
A new issue happened
[INFO] [launch]: All log files can be found below /home/hguo/.ros/log/2023-03-06-18-47-01-861177-hguo-Legion-5-Pro-7286
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('xml')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/frontend_launch_file_utilities.py", line 35, in get_launch_description_from_frontend_launch_file
    return parser.parse_description(root_entity)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/parser.py", line 111, in parse_description
    actions = [self.parse_action(child) for child in entity.children]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/parser.py", line 111, in <listcomp>
    actions = [self.parse_action(child) for child in entity.children]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/parser.py", line 88, in parse_action
    return instantiate_action(entity, self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/expose.py", line 39, in instantiate_action
    action_type, kwargs = action_parse_methods[entity.type_name](entity, parser)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/expose.py", line 105, in wrapper
    ret = found_parse_method(entity, parser)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/group_action.py", line 73, in parse
    kwargs['actions'] = [parser.parse_action(e) for e in entity.children]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/group_action.py", line 73, in <listcomp>
    kwargs['actions'] = [parser.parse_action(e) for e in entity.children]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/parser.py", line 88, in parse_action
    return instantiate_action(entity, self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/expose.py", line 38, in instantiate_action
    raise RuntimeError('Unrecognized entity of the type: {}'.format(entity.type_name))
RuntimeError: Unrecognized entity of the type: node_container

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 8 more times]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [xml]: Unrecognized entity of the type: node_container
Traceback (most recent call last):
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 61, in _tick_agent
    ego_action = self.agent()
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/leaderboard/autoagents/agent_wrapper.py", line 75, in __call__
[... cut ...]
Traceback (most recent call last):
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 171, in run_agent
    self.agent_loop._tick_agent(timestamp)
  File "/home/hguo/carla-autoware-universe/op_carla/op_bridge/op_bridge/op_bridge_ros2.py", line 67, in _tick_agent
    raise AgentError(e)
leaderboard.autoagents.agent_wrapper.AgentError: Stack exited with: 1 None
Scenario Ended , Hero has fallen, Sayonara
ERROR: failed to destroy actor 553 : unable to destroy actor: not found

yaoweixiao-ux, 2023-03-16:
I have same question

yaoweixiao-ux, 2023-03-17:
failed to destroy actor 553  git clone docker's openplanning at autoware/src/universe/exteral colcon build ageain

YimingChen-Real, 2023-03-28:
make sure in file xxx/op_agent/start_ros2.sh that map_path:=$HOME/xxx
refer to: [https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/#how-to-run-a-planning-simulation](url)

Desperadozcx, 2024-09-20:
> > I have same question
> 
> I figured it out by adding the command that got clone openplanner to the file named autoware.repos when following the instruction of autoware-galactic offficial instructions.

I encountered the same issue, and I have added the following content at line 28 in the autoware.repos file:

```
universe/external/open_planner:
    type: git
    url: https://github.com/ZATiTech/open_planner.git
    version: main
```

I have successfully downloaded open_planner, but I still encounter the problem you demonstrated. Do you know what the issue might be? What other modifications have you made?