hatem-darweesh/op_bridge #15: How to run ./run_exploration_mode_ros2.sh in autoware_universe docker

yaoweixiao-ux, 2023-03-07:
I want run  ./run_exploration_mode_ros2.sh in autoware.universe docker, outside dockers Error:No module named 'autoware_auto_vehicle_nsgs' in side docker can't find '/op_bridge/op_bridge_ros2.py' No such file or directory How Can I do now?

hatem-darweesh, 2023-03-08:
Sorry, I never used autoware inside docker. 
I won't be able to help you with that.

kshitiz-deloitte, 2023-04-07:
> I want run ./run_exploration_mode_ros2.sh in autoware.universe docker, outside dockers Error:No module named 'autoware_auto_vehicle_nsgs' in side docker can't find '/op_bridge/op_bridge_ros2.py' No such file or directory How Can I do now?

include LEADERBOARD_ROOT in bashrc of docker
This will solve the issue   '/op_bridge/op_bridge_ros2.py' No such file or directory

cfs4819, 2023-09-21:
`docker exec` or `docker run` command may not auto source  ~/.bashrc. So try copying the following code into ./run_exploration_mode_ros2.sh.
```sh
export CARLA_ROOT=path/to/carla-0.9.13
export SCENARIO_RUNNER_ROOT=path/to/op_carla/scenario_runner
export LEADERBOARD_ROOT=path/to/op_carla/op_bridge
export TEAM_CODE_ROOT=path/to/op_carla/op_agent
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/util
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/agents
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py3.7-linux-x86_64.egg
```
DO NOT forget to modify the paths into actual path in your container.