hatem-darweesh/op_bridge #16: my vehicle STATE always planning

yaoweixiao-ux, 2023-03-20:
when I set 2D pose Estimate point and 2D Goal Pose, vehicle can planning the line is yellow not green, error log failed look up transfrom from map to base_link;
[image]

yaoweixiao-ux, 2023-03-20:
I want to know how to solve this issus? 
my computer 20.04 galactic

hatem-darweesh, 2023-03-20:
it seems from screenshot the the localization is not working correctly. 
first you need to get stable localization. 
second. you need to check solution for this problem in another issues or openplanner-wg channel at autoware discord.

yaoweixiao-ux, 2023-03-21:
DId you means that my sensor config xml files have some error, I should deal with this error, and than ask autoware.universe issues ?

yaoweixiao-ux, 2023-03-21:
<launch>
  <arg name="launch_driver" default="true"/>
  <arg name="gnss_receiver" default="ublox" description="ublox(default) or septentrio"/>
  <arg name="coordinate_system" default="2" description="0:UTM, 1:MGRS, 2:PLANE"/>
  <arg name="plane_zone" default="0"/>

  <group>
    <push-ros-namespace namespace="gnss"/>

    
    <let name="navsatfix_topic_name" value="ublox/nav_sat_fix" if="$(eval &quot;'$(var gnss_receiver)'=='ublox'&quot;)"/>
    <let name="navsatfix_topic_name" value="septentrio/nav_sat_fix" if="$(eval &quot;'$(var gnss_receiver)'=='septentrio'&quot;)"/>
    <let name="orientation_topic_name" value="/autoware_orientation"/>

    
    <group if="$(eval &quot;'$(var gnss_receiver)'=='ublox'&quot;)">
      <node pkg="ublox_gps" name="ublox" exec="ublox_gps_node" if="$(var launch_driver)" respawn="true" respawn_delay="1.0">
        <remap from="~/fix" to="~/nav_sat_fix"/>
        <param from="$(find-pkg-share ublox_gps)/c94_f9p_rover.yaml"/>
      </node>
    </group>

    
    <group if="$(eval &quot;'$(var launch_driver)' and '$(var gnss_receiver)'=='septentrio'&quot;)">
      <include file="$(find-pkg-share septentrio_gnss_driver)/launch/mosaic_x5_rover.launch.xml"/>
    </group>

    
    <include file="$(find-pkg-share gnss_poser)/launch/gnss_poser.launch.xml">
      <arg name="input_topic_fix" value="$(var navsatfix_topic_name)"/>
      <arg name="input_topic_orientation" value="$(var orientation_topic_name)"/>

      <arg name="output_topic_gnss_pose" value="pose"/>
      <arg name="output_topic_gnss_pose_cov" value="pose_with_covariance"/>
      <arg name="output_topic_gnss_fixed" value="fixed"/>

      <arg name="coordinate_system" value="$(var coordinate_system)"/>
      <arg name="use_gnss_ins_orientation" value="true"/>

      <arg name="gnss_frame" value="gnss_link"/>
    </include>
  </group>
</launch>

This is my gnss.launch.xml I check it without find error

yaoweixiao-ux, 2023-03-22:
ERROR changed external_emergency_stop_heartbeat_received_time_ is false how can I deal with this problem? thank you

Eisoc, 2023-04-05:
> ERROR changed external_emergency_stop_heartbeat_received_time_ is false how can I deal with this problem? thank you

Hi, Did you solve this problem?

MarioAlmelaUMH, 2023-04-14:
Hello, I have a similar issue, but with some key differences in the console log. I've posted it here: https://github.com/hatem-darweesh/op_bridge/issues/19
in case someone can help me. Thanks in advance.

zeidk, 2023-06-11:
The issue is from `autoware.launch.xml`. In the youtube video tutorial some lines are missing.
I had some issues with localization and in my case I had to modify: `gnss.launch.xml`, `sensor_kit_calibration.yaml`, and `sensors_calibration.yaml`.

All these modified files are located here: https://drive.google.com/drive/folders/1_9l8P9oT6dhBDH0iMeJiGSPuBFy4owzr?usp=sharing

Make sure you make a backup of the old files before using the ones from the Google Drive.