hatem-darweesh/op_bridge #19: Ego Vehicle Stuck on Planning Stage

MarioAlmelaUMH, 2023-04-14:
Hello, I've been trying to run the bridge as shown in the video tutorials. I can input a 2D Goal correctly, but the ego vehicle remains stuck on Planning Stage, without ever engaging. 

[image]

Apparently, the vehicle plans part of the way correctly, but comes to a sudden stop, marked red in the trajectory preview. Additionally, the log obtained from the console is the following:

[system_error_monitor-21] [ERROR] [1681477470.405202187] [system_error_monitor /autoware/planning/performance_monitoring/trajectory_validation]: [Single Point Fault]: Stale
[system_error_monitor-21] [ERROR] [1681477470.405244454] [system_error_monitor /autoware/planning/performance_monitoring/trajectory_validation/trajectory_curvature_validation/planning_error_monitor: trajectory_curvature_validation]: [Single Point Fault]: This trajectory's curvature is within the expected range
[system_error_monitor-21] [ERROR] [1681477470.405257889] [system_error_monitor /autoware/planning/performance_monitoring/trajectory_validation/trajectory_point_validation/planning_error_monitor: trajectory_point_validation]: [Single Point Fault]: This Trajectory doesn't have any invalid values
[system_error_monitor-21] [ERROR] [1681477470.405268400] [system_error_monitor /autoware/planning/performance_monitoring/trajectory_validation/trajectory_interval_validation/planning_error_monitor: trajectory_interval_validation]: [Single Point Fault]: Trajectory Interval Length is within the expected range
[system_error_monitor-21] [ERROR] [1681477470.405277809] [system_error_monitor /autoware/planning/performance_monitoring/trajectory_validation/trajectory_relative_angle_validation/planning_error_monitor: trajectory_relative_angle_validation]: [Single Point Fault]: This trajectory's relative angle is within the expected range
[component_container_mt-64] [INFO] [1681477470.414272572] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: Replan with resetting optimization since valid nearest trajectory point from ego was not found.
[component_container-43] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[component_container-43] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[carla_pointcloud_node-1] CARLA CARLA >>>> Received Cloud : 0x7f5a94029e40, Converted: 
[component_container-67] [ERROR] [1681477472.544833715] [control.trajectory_follower.controller_node_exe]: [Emergency stop] vel: 0.000, acc: -5.000
[component_container-43] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[carla_pointcloud_node-1] CARLA CARLA >>>> Received Cloud : 0x7f5a94029e40, Converted: 
[component_container-67] [INFO] [1681477473.921484725] [control.operation_mode_transition_manager]: Engage unavailable: closest point not found
[component_container_mt-64] [INFO] [1681477473.930000145] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: Replan with resetting optimization since valid nearest trajectory point from ego was not found.
[component_container-43] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[carla_pointcloud_node-1] CARLA CARLA >>>> Received Cloud : 0x7f5a94029e40, Converted: 
[component_container-43] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[carla_pointcloud_node-1] CARLA CARLA >>>> Received Cloud : 0x7f5a94029e40, Converted: 
[component_container-43] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[lidar_centerpoint_node-47] [WARN] [1681477477.257861893] [lidar_centerpoint]: No detected boxes.
[component_container-67] [INFO] [1681477477.274596408] [control.operation_mode_transition_manager]: Engage unavailable: closest point not found

Is it PCL related, bridge related or Autoware related? I am using Ubuntu 20 with ROS 2 Galactic.

zeidk, 2023-06-11:
The issue is from `autoware.launch.xml`. In the youtube video tutorial some lines are missing.
I had some issues with localization and in my case I had to modify: `gnss.launch.xml`, `sensor_kit_calibration.yaml`, and `sensors_calibration.yaml`.

All these modified files are located here: https://drive.google.com/drive/folders/1_9l8P9oT6dhBDH0iMeJiGSPuBFy4owzr?usp=sharing

Make sure you make a backup of the old files before using the ones from the Google Drive.

MarioAlmelaUMH, 2023-06-12:
Thanks, I've tried it and now the vehicle advances to the Driving state:

[image]

However, the vehicle seems to find obstacles in its path. Those obstacles should not exist as the road is empty, there's no other vehicles nor pedestrians in the simulation yet. The vehicle also may appear slightly offset from its specified position, or slightly rotated.

[image]

I have tried to use your files, and also changed some sensor configuration values both in Autoware's sensor kit (sensor_kit_calibration.yaml) and in the bridge (op_ros2_agent.py), but the ego vehicle seems to always find obstacles in its path that don't exist. 

It is noticeable that this bug doesn't happen in specific maps and spawnpoints, but due to the nature of my work I have to be able to spawn the ego vehicle in multiple locations and scenarios without issue.

If anyone has any insight, it would be greatly appreciated.

Thanks in advance.

zeidk, 2023-06-14:
There is a clear all obects button in autoware. Can you try this and see if this helps? Based on the lidar visualization from your attached picture it does not seem that the lidar is detecting any object. This may come from something else but it is hard to tell without trying a few things.

zeidk, 2023-06-27:
Sorry for the late reply. Unfortunately I don't know what may cause this to happen. The slightly off rotation happens with me sometimes as well. I assume a more powerful computer may help here.

MarioAlmelaUMH, 2023-06-28:
Ok, I think I've solved it. The problem was that I tried to move the Lidar sensor to multiple positions besides the center (0, 0), alongside some differences between the bridge's config file and Autoware's. Once I corrected both aspects, the ego vehicle moves without problems in all maps.

Thank you for your answers.

closed 2023-06-28 by MarioAlmelaUMH

littlevu, 2024-01-15:
Hi @MarioAlmelaUMH, could you send me your sensor calibration positions in autoware and carla? Thank you very much!