hatem-darweesh/op_bridge #24: run_exploration_mode_ros2.sh, got error

NIDQ, 2024-01-10:
[gnss_poser-32] [ERROR] [1704860230.069107566] [sensing.gnss.gnss_poser]: Failed to convert Height from Ellipsoid to OrthometricFile not readable /usr/share/GeographicLib/geoids/egm2008-1.pgm
[gnss_poser-32] [ERROR] [1704860230.128319694] [sensing.gnss.gnss_poser]: Failed to convert Height from Ellipsoid to OrthometricFile not readable /usr/share/GeographicLib/geoids/egm2008-1.pgm
[gnss_poser-32] [ERROR] [1704860230.187034612] [sensing.gnss.gnss_poser]: Failed to convert Height from Ellipsoid to OrthometricFile not readable /usr/share/GeographicLib/geoids/egm2008-1.pgm
[gnss_poser-32] [ERROR] [1704860230.245079209] [sensing.gnss.gnss_poser]: Failed to convert Height from Ellipsoid to OrthometricFile not readable /usr/share/GeographicLib/geoids/egm2008-1.pgm
[gnss_poser-32] [ERROR] [1704860230.307171874] [sensing.gnss.gnss_poser]: Failed to convert Height from Ellipsoid to OrthometricFile not readable /usr/share/GeographicLib/geoids/egm2008-1.pgm
[gnss_poser-32] [ERROR] [1704860230.358996723] [sensing.gnss.gnss_poser]: Failed to convert Height from Ellipsoid to OrthometricFile not readable /usr/share/GeographicLib/geoids/egm2008-1.pgm
[component_container_mt-2] [WARN] [1704860229.378742897] [perception.occupancy_grid_map.occupancy_grid_map_node]: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[component_container_mt-2] [WARN] [1704860229.905567969] [perception.occupancy_grid_map.occupancy_grid_map_node]: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.

It seems that the coordinates are not aligned
Can anyone provide some help？

NIDQ, 2024-01-11:
New error!

[imu_corrector-31] [WARN] [1704941917.759844572] [sensing.imu.imu_corrector]: failed to get transform from tamagawa/imu_link to base_link: "tamagawa/imu_link" passed to lookupTransform argument target_frame does not exist. 
[imu_corrector-31] [ERROR] [1704941917.759985300] [sensing.imu.imu_corrector]: Please publish TF base_link to tamagawa/imu_link
[ndt_scan_matcher-37] [INFO] [1704941917.752543819] [localization.pose_estimator.ndt_scan_matcher]: points_queue_size: 1
[ndt_scan_matcher-37] [INFO] [1704941917.752621838] [localization.pose_estimator.ndt_scan_matcher]: base_frame_id: base_link
[ndt_scan_matcher-37] [INFO] [1704941917.752634956] [localization.pose_estimator.ndt_scan_matcher]: ndt_base_frame_id: ndt_base_link
[ndt_scan_matcher-37] [INFO] [1704941917.752681973] [localization.pose_estimator.ndt_scan_matcher]: trans_epsilon: 0.010000, step_size: 0.100000, resolution: 2.000000, max_iterations: 30
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/system/system_monitor/mem_monitor' in container '/system/system_monitor/system_monitor/system_monitor_container'
[ekf_localizer-39] [WARN] [1704941917.762371596] [localization.pose_twist_fusion_filter.ekf_localizer]: The node is not activated. Provide initial pose to pose_initializer
[component_container-28] [INFO] [1704941917.792446045] [sensing.lidar.left.velodyne_convert_node]: This node is only tested for VLP16, VLP32C, and VLS128. Use other models at your own risk.
[component_container-28] [INFO] [1704941917.795439268] [sensing.lidar.left.velodyne_convert_node]: correction angles: /home/p30031852276/app/carla-autoware-universe/autoware/install/velodyne_pointcloud/share/velodyne_pointcloud/params/VLP16db.yaml
[component_container-28] [INFO] [1704941917.795495314] [sensing.lidar.left.velodyne_convert_node]: correction angles: /home/p30031852276/app/carla-autoware-universe/autoware/install/velodyne_pointcloud/share/velodyne_pointcloud/params/VLP16db.yaml
[component_container-28] [INFO] [1704941917.799784182] [calibration]: laser_ring[ 0] =  0, angle = -0.261799
[component_container-28] [INFO] [1704941917.799806935] [calibration]: laser_ring[ 2] =  1, angle = -0.226893
[component_container-28] [INFO] [1704941917.799809942] [calibration]: laser_ring[ 4] =  2, angle = -0.191986
[component_container-28] [INFO] [1704941917.799812173] [calibration]: laser_ring[ 6] =  3, angle = -0.157080
[component_container-28] [INFO] [1704941917.799814407] [calibration]: laser_ring[ 8] =  4, angle = -0.122173

hatem-darweesh, 2024-01-11:
I need more information to know what is wrong. 
what are the detailed steps ? 
have you put the maps in one folder correctly ?