hatem-darweesh/op_bridge #25: Multiple Lidar Support

mraditya01, 2024-01-13:
Hello!

Thank you very much for your amazing work of Autoware CARLA bridging, it works really well for me. I have some question when adding one more LIDAR to the system.
I have tried to add one more LIDAR to the system by modifying some code,

- agent_wrapper.py Line 27

`SENSOR_LIMITS = {
...
'sensor.lidar.ray_cast': 2,
...`

- op_ros2_agent.py Line 316 (Adding more lidar.ray_cast with different ID here)

`def sensor(self): sensors = ...`

- carla_pointcloud_interface_node.cpp and .hpp (Added one more subscriber and publisher for the new Lidar Nodes)

After modifying those codes, the localization fail to run as only one lidar topic have message (ex. carla_pointcloud_1, where as carla_pointcloud_2 is empty) and both the publisher topic of the lidar message for Autoware are empty somehow. Do you have work around or an idea on how to add more LIDAR to your method?

Thank you in advance.

hatem-darweesh, 2024-01-13:
Hi @mraditya01 
This is challenging. I haven't tried it before. 
But I will share with you two comments that should make your life easier. 
1- CARLA side: Do both LIDARs publish data correctly ? If yes then that is great and go to the second step. 

2- When you send multiple Lidar to Autoware you need to know which one will be used in Localization and which will be used in Perception. Some of the Lidars are combined in one cloud others not. It is not clear in the documentation how to achieve that. 

As a work around to avoid all that, I recommend to combine both LIDARs in the bridge side and publish on Pointcloud topic to autoware.

hatem-darweesh, 2024-01-13:
@mraditya01 also you can contact me using the Lab Slack, I am still active there.