hatem-darweesh/op_bridge #26: The initialization of localization is very poor in the ROS Humble version

datvuthanh, 2024-01-23:
Dear @hatem-darweesh ,

I have successfully run your Carla Autoware bridge in the ROS Humble version. However, the initialization of localization seems to be quite poor. 

I am not sure why; usually, I have to rerun the program several times to get the localization initialization right.

[image]

Thank you very much!

cfs4819, 2024-01-23:
It happens to me also. 

So I try respawning the vehicle for 2 or 3 times. After that it may works fine.  That is also why I pull request "[make it possible to publish initialpose via rviz](https://github.com/hatem-darweesh/op_bridge/pull/23)", which helps me respawning vehicle more easily.

There is no other way I can try.

datvuthanh, 2024-01-23:
Thank you very much!

datvuthanh, 2024-01-23:
Hi @cfs4819 , could you run successfully run_route_scenarios.sh?

cfs4819, 2024-01-23:
> Hi @cfs4819 , could you run successfully run_route_scenarios.sh?

Sorry, I haven’t tried run_route_scenarios.sh before.

hatem-darweesh, 2024-01-23:
It happens with me too.  

There are 2 reasons: 
1- Computation load is too high for the PC
2- The point cloud is not that good, actually it is a bit different than the actual simulation environment. 

New pointcloud map + HD road network map (osm) need to be created for CARLA environments.