hatem-darweesh/op_bridge #27: Localization stuck on "uninitialized"

dtonda8, 2024-01-27:
Hey @hatem-darweesh. Thank you for providing a Carla bridge that is compatible with Autoware. Please help me with this issue.

Here's what I've done so far:
My system:
Ubuntu 22.04
Python 
Carla 0.9.15

I followed the steps listed on [op_carla_bridge](https://github.com/ZATiTech/open_planner/tree/humble/op_carla_bridge) and cloned [scenario_runner (openplanner_carla_bridge branch)](https://github.com/hatem-darweesh/scenario_runner/tree/openplanner_carla_bridge). Here is what some of my directories look like: 
[image]

I had errors when I placed `carla_sensor_kit` directly in `individual_params`. 
[image]

I managed to get around this error by placing  `carla_sensor_kit` in `individual/config/default` instead:
[image]

`colcon build` build initially had some `stderr`s on packages. But after trying again it successfully built:
[image]

I created a directory called `carla-0.9.15`, extracted the `CARLA_0.9.15.tar.gz` into that directory. I also installed `op_agent` and `op_bridge` into that directory, so it looks like this:
[image]

Finally I tried following the steps 1-2 in [op_bridge](https://github.com/hatem-darweesh/op_bridge/tree/ros2-humble):
>How to run the bridge:
>1-Run CARLA simulator 0.9.13
>2-Run one of the following options.
>3-In case of run_srunner_agent.sh, scneario must be started before running the bridge.
>4-The bridge run script will run the op_agent

First Terminal: `./dbat/carla-0.9.15/CarlaUE4.sh `

Second Terminal:
Some additionally `export`s I had to make not mentioned in op_bridge/op_agent repos
```
export OP_BRIDGE_ROOT=~/dbat/carla-0.9.15/op_bridge
export OP_AGENT_ROOT=~/dbat/carla-0.9.15/op_agent
export OP_BRIDGE_ROOT=~/dbat/carla-0.9.15/op_bridge
export AUTOWARE_ROOT=~/dbat/autoware
```

Then run the rest of the `export`s listed in op_bridge/op_agent repos
```
export CARLA_ROOT=~/dbat/carla-0.9.15
export SCENARIO_RUNNER_ROOT=~/dbat/carla-0.9.15/scenario_runner
export LEADERBOARD_ROOT=~/dbat/carla-0.9.15/op_bridge/leaderboard 
export TEAM_CODE_ROOT=~/dbat/carla-0.9.15/op_agent 
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/util 
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla 
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/agents
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py2.7-linux-x86_64.egg
```

Finally: `./dbat/carla-0.9.15/op_bridge/op_scripts/run_exploration_mode_ros2.sh`

Output:
https://github.com/hatem-darweesh/op_bridge/assets/99521514/469bee2c-7f7c-454e-b622-d9eafc122e03

State of car after several minutes:
[image]

hatem-darweesh, 2024-01-27:
Can you check: 
1- 
"export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.13-py2.7-linux-x86_64.egg" 
you need to download the python egg for 3.10 from the mentioned link. and change the above export to the new egg name. 

2- Download the Pointcloud PCD map and create a map folder contains both .osm and .pcd files. 

I recommend you familiarize yourself with autoware.universe fist by running both the planning and rosbag simulation mentioned in the documentation.

dtonda8, 2024-01-27:
Thanks for responding.
1. You helped me realise that this export does nothing, I don't have carla-0.9.13. It should have looked like this: `export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.15-py3.7-linux-x86_64.egg`

This export causes this error:
[image]

The issue is that I can't seem to find the python egg for 3.10. The latest version I could find is the egg file for 3.7. If anyone has found it for 3.10 please let me know.

2. I'll will this suggestion

hatem-darweesh, 2024-01-27:
You need to follow the instruction from the "humble" branch 

https://github.com/hatem-darweesh/op_bridge/blob/ros2-humble/README.md

dtonda8, 2024-01-27:
Thanks for providing the link for the Python3.10 egg file. The homepage of op_bridge repo shows a different README.md than the one you linked above. I think that's why I couldn't find this egg file.

I retried again using `export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.15-py3.10-linux-x86_64.egg` and getting similar errors as before:
[image]
Most of the errors are related to `base_link` and `tf`

dtonda8, 2024-01-28:
I manage to partially resolve the issue above.
```
[map.map_hash_generator]: [/home/dbat/carla-0.9.15/op_agent/autoware-contents/maps/Town01/pointcloud_map.pcd] is neither file nor directory
```
Downloaded pcd map from [carla simulator maps](https://bitbucket.org/carla-simulator/autoware-contents/src/master/maps/point_cloud_maps/Town01.pcd), moved it to the location it was searching for and renamed to `pointcloud_map.pcd`

```
1706412378.0583673 [web_server.py-75] Traceback (most recent call last):
1706412378.0591691 [web_server.py-75]   File "/home/mcav/dbat/autoware/install/default_ad_api/lib/default_ad_api/web_server.py", line 21, in <module>
1706412378.0600290 [web_server.py-75]     import flask
1706412378.0893207 [web_server.py-75]   File "/usr/lib/python3/dist-packages/flask/__init__.py", line 7, in <module>
1706412378.0937130 [web_server.py-75]     from .app import Flask as Flask
1706412378.1007218 [web_server.py-75]   File "/usr/lib/python3/dist-packages/flask/app.py", line 28, in <module>
1706412378.1114581 [web_server.py-75]     from . import cli
1706412378.1141617 [web_server.py-75]   File "/usr/lib/python3/dist-packages/flask/cli.py", line 18, in <module>
1706412378.1183803 [web_server.py-75]     from .helpers import get_debug_flag
1706412378.1305094 [web_server.py-75]   File "/usr/lib/python3/dist-packages/flask/helpers.py", line 16, in <module>
1706412378.1308119 [web_server.py-75]     from werkzeug.urls import url_quote
1706412378.1333606 [web_server.py-75] ImportError: cannot import name 'url_quote' from 'werkzeug.urls' (/home/mcav/.local/lib/python3.10/site-packages/werkzeug/urls.py)
```
 `pip install Werkzeug==2.2.0` fixed it.

I am able to see the map and set '2D goal pose'. :)

[image]

The only issue now is that I sometimes need to rerun the program since initialization is inconsistent. Relevant errors:
```
1706417562.2970629 [pose_initializer_node-36] [WARN] [1706417562.296250151] [localization.util.pose_initializer_node]: failed to lookup transform: Could not find a connection between 'map' and 'gnss_link' because they are not part of the same tree.Tf has two or more unconnected trees.
1706417562.2986362 [pose_initializer_node-36] [INFO] [1706417562.296862493] [localization.util.pose_initializer_node]: Call NDT align server.
1706417562.3000901 [ndt_scan_matcher-35] [WARN] [1706417562.299047845] [localization.pose_estimator.ndt_scan_matcher]: Could not find a connection between 'map' and 'gnss_link' because they are not part of the same tree.Tf has two or more unconnected trees.
1706417562.3002465 [ndt_scan_matcher-35] [ERROR] [1706417562.299101175] [localization.pose_estimator.ndt_scan_matcher]: Please publish TF map to gnss_link
```

I'd love to be able to at least move the car with '2D Pose Estimate'.

dtonda8, 2024-01-28:
After adding two files from this [pr](https://github.com/hatem-darweesh/op_bridge/pull/23) by @cfs4819, I can now successfully use '2D Pose Estimate' .

Thanks Hatem-darweesh and cfs4819

closed 2024-01-28 by dtonda8