hatem-darweesh/op_bridge #28: Missing Required Argument 'component_state_monitor_topic_path' in run_exploration_mode_ros2.sh

AmirhoseinCh, 2024-02-29:
Hi,
I encountered an error while running the ./run_exploration_mode_ros2.sh script inside the autoware-universe:humble-2023.10-cuda Docker image. The error message is as follows:
`[ERROR] [launch]: Caught exception in launch (see debug for traceback): Included launch description missing required argument 'component_state_monitor_topic_path' (description: 'no description given'), given: [map_path, vehicle_model, sensor_model, run_mode, sensor_model, cpu_monitor_config_file, gpu_monitor_config_file, hdd_monitor_config_file, mem_monitor_config_file, net_monitor_config_file, ntp_monitor_config_file, process_monitor_config_file, voltage_monitor_config_file, mode, file, config_file, extra_agg_config_file_sensing, extra_agg_config_file_system, extra_agg_config_file_vehicle, use_emergency_hold, config_file, ignore_missing_diagnostics, add_leaf_diagnostics, data_ready_timeout, data_heartbeat_timeout, diag_timeout_sec, hazard_recovery_timeout, use_emergency_hold, use_emergency_hold_in_manual_driving, config_file, config_file, config_file, config_file, extra_config_file_sensor, launch_rqt_reconfigure, launch_rqt_runtime_monitor, launch_rqt_robot_monitor, launch_rqt_runtime_monitor_err, config_file, extra_config_file_sensor, update_rate]
Traceback (most recent call last):
  File "/home/amir/Carla/CARLA_0.9.15/op_bridge-ros2-humble/op_bridge/op_bridge_ros2.py", line 61, in _tick_agent
    ego_action = self.agent()
  File "/home/amir/Carla/CARLA_0.9.15/op_bridge-ros2-humble/leaderboard/autoagents/agent_wrapper.py", line 75, in __call__
    return self._agent()
  File "/home/amir/Carla/CARLA_0.9.15/op_bridge-ros2-humble/leaderboard/autoagents/autonomous_agent.py", line 115, in __call__
    control = self.run_step(input_data, timestamp)
  File "/home/amir/Carla/CARLA_0.9.15/op_bridge-ros2-humble/op_bridge/op_ros2_agent.py", line 528, in run_step
    raise RuntimeError("Stack exited with: {} {}".format(
RuntimeError: Stack exited with: 1 None
`
The environment details are as follows:

Docker image: autoware-universe:humble-2023.10-cuda
Carla version: 0.9.15
The op_bridge and op_agent are cloned in carla directory. I've cloned the open_planer and Lidar driver and rebuilt autoware, but the error persists.

Could you please assist in resolving this issue?

hatem-darweesh, 2024-02-29:
1. I didn't test autoware with with Docker. 
2. This error could mean many things. 
3. Check you have the "humbler" branch for op_bridge and op_agent. 
4. Check that autoware work separately using the planning and rosbag replay simulations. 
5. Check the CARLA work with op_bridge separately:
     - go to op_agent/start_ros2.sh
     - comment the line "ros2 launch ${OP_AGENT_ROOT}/autoware_carla_launch/carla_simulator.launch.xml map_path:=${OP_AGENT_ROOT}/autoware-contents/maps/$map_name vehicle_model:=sample_vehicle sensor_model:=carla_sensor_kit" 
     - un-comment the line "# $SHELL"
     - run carla simulator 
     - run "./run_exploration_mode_ros2.sh"

AmirhoseinCh, 2024-02-29:
Thanks for your reply.

The op_bridge and op_agent are cloned from the humble branch. The autoware works fine separately in rosbag replay simulation.
The op_bridge works separately, I attached the output.

[image]

 However, the error I mentioned before persist. Here is how I launch it in the "./run_exploration_mode_ros2.sh":
 `ros2 launch /home/amir/autoware/src/universe/external/open_planner/op_launch/autoware-carla.launch.xml map_path:=${OP_AGENT_ROOT}/autoware-contents/maps/$map_name vehicle_model:=sample_vehicle    sensor_model:=carla_sensor_kit`

TimDing-aoi, 2025-03-07:
Did anyone manage to solve this issue?

AOOOOOA, 2025-06-25:
> Did anyone manage to solve this issue?

Hi Have you solved this problem?

TimDing-aoi, 2025-08-04:
> > Did anyone manage to solve this issue?
> 
> Hi Have you solved this problem?

No, but I did manage to get run exploration mode running, under a new python environment