hatem-darweesh/op_bridge #30: How to run CARLA leaderboard routes with Auoware.universe and openplanner on ROS 2 (galactic)?

ry4nzhu, 2024-04-24:
Hi,

Thanks for the great code and tutorial. I tried to setup autoware.universe with the Carla bridge on Ubuntu 20.04 and ROS galactic. Now I can successfully run the `run_exploration_mode_ros2.sh` to play with the vehicle planning. 

However, when I try to run `run_route_scenarios_ros2.sh` on the Leaderboard routes, the ego vehicle is always stuck at the waiting_for_route state and does not follow the predefined waypoints in the route. It looks to me that the vehicle is not receiving any routing points to do the planning, but I am not sure what causes this.

[image]

Do you know which part of the code could possibly go wrong here?

hatem-darweesh, 2024-05-07:
Sorry the leader board is not integrated with ROS2 op_bridge , it only tested with ROS1 and autoware.ai version. 
you will need to follow check the ROS1 bridge and modify the ROS2 bridge accordingly.

ry4nzhu, 2024-05-07:
Thanks for your reply. I am trying to modify the ROS2 bridge to run the leaderboard. I have one question: how to redirect the `external_route_topic` (i.e. the waypoints generated from the leaderboard scripts) to Autoware so that the planning can start automatically (instead of in exploration mode we have to set the goal pose manually)?

Can you provide me with some guidance on which part I should look into?